ewpa/LibSSH-ESP32 #1: Undefined reference to `ssh_config_parse_line'

Acipenser, 2020-07-08:
Hi,

in order to let the example code run, I entered my WiFi credentials in the client example code in line [12 and 13](https://github.com/ewpa/LibSSH-ESP32/blob/fa1a4e7f66a348b9271eb15a664b669768c3de32/examples/exec/exec.ino#L12) and my SSH credentials in line [494](https://github.com/ewpa/LibSSH-ESP32/blob/fa1a4e7f66a348b9271eb15a664b669768c3de32/examples/exec/exec.ino#L494).

While uploading, the following errors show up:
```
libraries/LibSSH-ESP32/LibSSH-ESP32.a(config.c.o):(.literal.ssh_config_parse_file+0xc): undefined reference to `ssh_config_parse_line'
libraries/LibSSH-ESP32/LibSSH-ESP32.a(config.c.o): In function `ssh_config_parse_file':
/Users/Acipenser/Documents/Arduino/libraries/LibSSH-ESP32/src/config.c:1164: undefined reference to `ssh_config_parse_line'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module.
```

Did I miss something here?

Thanks for your help!

ewpa, 2020-07-08:
Hi,

It appears that with commit fa1a4e7f I removed the implementation of `ssh_config_parse_line`.  Please can you retry as before using the earlier release 1.1.1.  If this fixes the issue then I will implement a fix for the later release.

Thank you for your report.

Acipenser, 2020-07-08:
Can confirm, with 1.1.1 error-free compiling!

ewpa, 2020-07-08:
This is fixed in commit 7c4439b and released as 1.1.3.  No real need to update if you are happy with release 1.1.1.

closed 2020-07-08 by ewpa