ewpa/LibSSH-ESP32 #2: Error opening "/spiffs/.ssh/id_ed25519" file

Acipenser, 2020-07-09:
Hi, me again :) 

While executing the unaltered keygen2 example, the serial monitor showed the following output:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
% Mounted SPIFFS used=1506 total=1378241

% Execution in progress: keygen2 --type ed25519 --file /spiffs/.ssh/id_ed25519

Error opening "/spiffs/.ssh/id_ed25519" file

% Execution completed: rc=-1
```

Any suggestions on that? 
Thanks for your help!

ewpa, 2020-07-09:
Hello, that example cowardly refuses to overwrite the key file if it exists already.  I believe your ESP32 had run the sketch immediately after upload and before you connected to the console and collected this output.  To confirm this I erased an ESP32, uploaded the example, and ran it twice.

The first execution, I forced a format in the code to wipe spiffs.

```
E (54) SPIFFS: mount failed, -10025
% No formatted SPIFFS filesystem found to mount.
% Format SPIFFS and mount now (NB. may cause data loss) [y/n]?
% Formatting...
% Mounted SPIFFS used=0 total=1374476
% Execution in progress: keygen2 --type ed25519 --file /spiffs/.ssh/id_ed25519

% Execution completed: rc=0
```

Then I reset the ESP32 and saw the same message as you, noting the spiffs has data before this execution.

```
% Mounted SPIFFS used=1255 total=1374476
% Execution in progress: keygen2 --type ed25519 --file /spiffs/.ssh/id_ed25519

Error opening "/spiffs/.ssh/id_ed25519" file

% Execution completed: rc=-1
```

So, for the solution it is simplest to change the file name in the code and do not let your device reset before connecting to the console.

Acipenser, 2020-07-10:
Thanks for the clarification, that helped!

closed 2020-07-10 by Acipenser