ewpa/LibSSH-ESP32 #3: keygen2 for rsa, dsa, ecdsa

nikolaynazarko, 2020-07-14:
Hi. Thanks a lot for your efforts in this project. 
I just want to ask is keygen2 support rsa, dsa, ecdsa key types? 
Because when I try RSA I got  id_rsa file with 0 size(I've added some code for browse files ):

[image]

Same results for dsa, ecdsa. Only ed25519 working on my side.
Best regards.

ewpa, 2020-07-14:
Hi, I looked at this for some time in the past while trying to create a host key directly.  The combination of libssh and mbedtls do not have the implementations needed to export the keys other than for the ed25519 type.  To confirm, I built this port on Linux using the same build configuration and get the same error.
```
$ ./examples/keygen2 --type rsa --bits 1024 --file ./id_rsa
Error: Failed to write private key file
```
So this is an existing limitation and there is no fix in scope.

nikolaynazarko, 2020-07-14:
Ok. Thank you.

closed 2020-07-14 by nikolaynazarko