ewpa/LibSSH-ESP32 #4: RSA private encrypt and public decrypt

radimkohout, 2020-08-29:
Hello,
does this library support encrypting with private RSA key and decrypting with public RSA key? If so, how to do it?
Thanks

ewpa, 2020-08-29:
This is really a question for upstream.  However I would suggest you consider using the existing mbedtls library on the ESP32 to achieve this, e.g. [How to encrypt and decrypt with RSA - Knowledge Base - Mbed TLS](https://tls.mbed.org/kb/how-to/encrypt-and-decrypt-with-rsa).

radimkohout, 2020-08-29:
I can't figure out how to do it with mbedTLS.

ewpa, 2020-08-30:
Where are you having issues?  Have you created a sketch that compiles and runs?  I searched the libssh PKI API and could not find the calls you would need for RSA encryption or decryption, so on the face of it, libssh does not support that functionality.

radimkohout, 2020-08-31:
The problem is, as you stated, that libssh doesn't support these functions

ewpa, 2020-08-31:
I can say that the mbed link I provided does do what you need: encrypt and decrypt.  I've used something similar successfully.

radimkohout, 2020-08-31:
But with reversed keys than normally?

ewpa, 2020-08-31:
What do you mean?

radimkohout, 2020-09-02:
Normally you encrypt with public and decrypt with private key. I need to encrypt with private and decrypt with public.

ewpa, 2020-09-03:
I understand now.  I do not believe that is possible unless you are 'signing'.

closed 2021-11-07 by ewpa