ewpa/LibSSH-ESP32 #5: Exception after few connect\disconnect.

nikolaynazarko, 2020-09-24:
Hello. I found some exception after a few connect\disconnect to\fom esp32 ssh server.

`Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060430  A0      : 0x801002ff  A1      : 0x3ffb1d00  
A2      : 0x3ffc6cac  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffeed54  A8      : 0x80122e89  A9      : 0x3ffb1cf0  
A10     : 0x00000000  A11     : 0x3ffb1d0c  A12     : 0x00000000  A13     : 0x3ffb1e8c  
A14     : 0x0000003b  A15     : 0x00000018  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb1d00 0x401002fc:0x3ffb1eb0 0x400fc496:0x3ffb1ed0 0x400fc508:0x3ffb1f00 0x400ec9f4:0x3ffb1f20 0x400d6ddb:0x3ffb1f90 0x4010e0a5:0x3ffb1fb0 0x40089fe1:0x3ffb1fd0

Rebooting...
`
And decoded:

`PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x401002fc: ssh_reseed at C:\Users\kalyan\Documents\Arduino\libraries\LibSSH-ESP32\src\libmbedcrypto.c line 50
0x400fc496: ssh_bind_accept_fd at C:\Users\kalyan\Documents\Arduino\libraries\LibSSH-ESP32\src\bind.c line 549
0x400fc508: ssh_bind_accept at C:\Users\kalyan\Documents\Arduino\libraries\LibSSH-ESP32\src\bind.c line 574
0x400ec9f4: ssh_serv_proc() at C:\Users\kalyan\AppData\Local\Temp\arduino_build_582532\sketch\ssh_server.cpp line 417
0x400d6ddb: loop() at D:\my_projects\sshESP32\sshESP32.ino line 137
0x4010e0a5: loopTask(void*) at C:\Users\kalyan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
0x40089fe1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
`
NOTE: After the client is disconnect from the server I'm free all server resources and setup server again to be able to accept a new connection.

Maybe you have some ideas about it.

Thank you. Best regards

ewpa, 2020-09-24:
The stack size for the "loop" task is probably too small.  In the examples I create a "ctl" task with a larger stack.  20 kB will probably be large enough.  I free messages, channels, sessions and keys, but nothing else in my own server code.  This is to avoid fragmentation with such a small memory space in the ESP32.  Let me know if making those changes improves your code stability.

nikolaynazarko, 2020-09-25:
Seems task heap size is not a problem. The same exception even if the task heap size is 30 Kb. Here are the logs:

`Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060630  A0      : 0x800ec303  A1      : 0x3ffeb840  
A2      : 0x3ffc6b08  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffdfc50  A8      : 0x8010d0c9  A9      : 0x3ffeb830  
A10     : 0x00000000  A11     : 0x3ffeb84c  A12     : 0x00000000  A13     : 0x3ffeb9cc  
A14     : 0x0000003b  A15     : 0x00000018  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffeb840 0x400ec300:0x3ffeb9f0 0x400e84a6:0x3ffeba10 0x400e8518:0x3ffeba40 0x400dd018:0x3ffeba60 0x400dd536:0x3ffebad0 0x40088b9d:0x3ffebaf0

Rebooting...
`
Decoded:

`PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x400ec300: ssh_reseed at C:\Users\kalyan\Documents\Arduino\libraries\LibSSH-ESP32\src\libmbedcrypto.c line 50
0x400e84a6: ssh_bind_accept_fd at C:\Users\kalyan\Documents\Arduino\libraries\LibSSH-ESP32\src\bind.c line 549
0x400e8518: ssh_bind_accept at C:\Users\kalyan\Documents\Arduino\libraries\LibSSH-ESP32\src\bind.c line 576
0x400dd018: ssh_serv_proc() at C:\Users\kalyan\AppData\Local\Temp\arduino_build_943769\sketch\ssh_server.cpp line 413
0x400dd536: ssh_serv_thread(void*) at C:\Users\kalyan\AppData\Local\Temp\arduino_build_943769\sketch\ssh_server.cpp line 306
0x40088b9d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
`

ewpa, 2020-09-25:
Nothing much going on at libmbedcrypto.c line 50: mbedtls_ctr_drbg_reseed(&ssh_mbedtls_ctr_drbg, NULL, 0);
I wonder what is in ssh_mbedtls_ctr_drbg at the time of the error.  Can you capture with gdb or other means?  Has all ssh initialization been called before the bind attempt?  Variable ssh_mbedtls_ctr_drbg should be initialized in ssh_crypto_init, called from _ssh_init, called from libssh_constructor.  Since libssh is not compiled as a 'library' in the normal sense we have to call the constructor manually.  This is achieved by calling libssh_begin as per the README.md:

```
#include "libssh_esp32.h"
libssh_begin();
```

Please verify this is being called before any other ssh calls.

nikolaynazarko, 2020-09-25:
Well, my session clean-up code was wrong. It was like this:

`  ssh_channel_close(chan);

   ssh_disconnect(session);

   ssh_bind_free(sshbind);

   ssh_free(session);

   ssh_finalize();`

But seems ssh_finalize() destruct libssh and need call libssh_begin() again.
So, without ssh_finalize() working stable. I can connect\disconnect without problems now.
Thank you for the cool lib!

closed 2020-09-25 by ewpa