ewpa/LibSSH-ESP32 #6: SSHD Working example

ShahoffAndrew, 2020-10-22:
Is it there some working step-by-step/how-to manual for keygen/sshd ?

I am trying to generate rsa and dsa with keygen2 , but getting 0 filelenght. Only ed25519 key file and public file can be created.
But than how to start  sample sshd server wit this ed25519 key file ?

Any help will be much appreciated 

/Andrew

ewpa, 2020-10-22:
If you need the end-to-end workflow on the ESP32 then you are limited to ed25519.  Please refer to  issue#3 for further information.  Can you make the following change to samplesshd-kbdint below and report on your results?  The change should allow the SSH server example to work with the key created earlier in keygen2.

On line 28 of _samplesshd-kbdint.ino_, replace
`#define EX_CMD "samplesshd-kbdint", "::"`
with
`#define EX_CMD "samplesshd-kbdint", "--hostkey", "/spiffs/.ssh/id_ed25519", "::"`

ewpa, 2020-12-28:
Fixed issue #6 with new release 1.2.3 as commit e15200e.

closed 2020-12-28 by ewpa