ewpa/LibSSH-ESP32 #8: libssh argp_parse error (not declared in this scope) while porting libssh from arduino to esp-idf

youssefIOT, 2021-06-02:
i get the following error when compiling under esp-idf
i have all argp files (.c ans .h) in the main project directory. thank for help !!
---------------------------------------------------------------------------------------------------------------------
../main/SSH_TO_IDF.cpp: In function 'int ex_main(int, const char**, int)':
../main/SSH_TO_IDF.cpp:1199:5: error: 'argp_parse' was not declared in this scope
     argp_parse(&argp, argc,  argv, 0, 0, sshbind);
     ^~~~~~~~~~
../main/SSH_TO_IDF.cpp:1199:5: note: suggested alternative: 'argp_parser_t'
     argp_parse(&argp, argc,  argv, 0, 0, sshbind);
     ^~~~~~~~~~
     argp_parser_t

ewpa, 2021-06-02:
I have not tested the library as an Arduino component of IDF, just under the Arduino IDE and Arduino Builder.  I would need to set up an environment before attempting to recreate the issue.  I note above the first error is in the compiler and the second error in the linker.  Unless I can recreate I am unable to be sure of the significance.  If possible could you upload a test IDF project to GitHub with this library as a submodule.

ewpa, 2024-09-28:
Fixed with commit 057ada82.  Please check and confirm.