ewpa/LibSSH-ESP32 #9: Is it possible to establish SSH connection while using softAP ?

GeorgeFlorian, 2021-07-19:
I am interested in establishing a SSH connection with a server.
I want the server the be connected to the ESP using `softAP()`.
Is this possible ?

I've went to the https://github.com/ewpa/LibSSH-ESP32/tree/master/examples folder but I have no idea what to use from there.

One thing to point out from the example files is that there are a lot of duplicate includes.
For example in exec example:
```
#include "examples_common.h" // is declared 4 times
#include <libssh/libssh.h> // is declared 4 times
#include <stdio.h> // is declared 3 times
// and so on
```

ewpa, 2021-07-19:
Hi George, is your requirement that the server will get its IP address from the ESP32 and then the ESP32 makes an SSH connection to the server?  Or that the ESP32 is a server and can be reached from connecting devices as an AP?

The multiple includes are a result of porting the example code from upstream 'as-is'.

GeorgeFlorian, 2021-07-19:
> Hi George, is your requirement that the server will get its IP address from the ESP32 and then the ESP32 makes an SSH connection to the server? Or that the ESP32 is a server and can be reached from connecting devices as an AP?
>

Hello.
Thank you for  your prompt reply.

The PC server will be connected to a network.
Also, the same PC server should connect to the ESP32's AP on Wireless to enable communication between them.

I am trying to connect to the ESP using softAP to try and eliminate the need for a switch between the PC Server and the ESP32.

What do you think ?

> The multiple includes are a result of porting the example code from upstream 'as-is'.

Got it. Thanks.
I am not very good at networking so can you please help me with some answers ?

Inside `exec` example there are multiple functions doing different things:
```
/*
 * knownhosts.c
 * This file contains an example of how verify the identity of a
 * SSH server using libssh
 */
int verify_knownhost(ssh_session session):

/*
 * authentication.c
 * This file contains an example of how to do an authentication to a
 * SSH server using libssh
 */
int authenticate_kbdint(ssh_session session, const char *password):
static int auth_keyfile(ssh_session session, char* keyfile);
int authenticate_console(ssh_session session)

/*
 * connect_ssh.c
 * This file contains an example of how to connect to a
 * SSH server using libssh
 */
ssh_session connect_ssh(const char *host, const char *user,int verbosity):
int ex_main(int argc, char **argv);
```

There are basically 3 examples doing 3 different things.
To establish the type of connection that I need, what functions do I have to use ?

ewpa, 2021-07-19:
The examples to try to run the ESP32 as an SSH server are firstly `keygen2`.  This will create the SSH host key on the ESP32.  Secondly, `samplesshd-kbdint.`  This runs a sample SSH server on the ESP32.  Once this is working that sketch can be enhanced to use AP mode.  Have a look through those examples first, especially the second example which runs the server.

GeorgeFlorian, 2021-07-20:
> The examples to try to run the ESP32 as an SSH server are firstly `keygen2`. This will create the SSH host key on the ESP32. Secondly, `samplesshd-kbdint.` This runs a sample SSH server on the [...]

I don't think that I need the ESP32 to run a SSH Server. I just need the ESP to send one command to the PC using SSH.
How should I breakdown the code for this ?
What is the `client code` ?
After the `client code` comes the `client's private key`, right ?

ewpa, 2021-07-20:
Yes, seems `exec` example is the closest to what you need to SSH to your server that connects to the ESP32 as AP.  The `verify_knownhost` ensures you connect to only a server you already know.  `auth_keyfile` is probably the best way to authenticate.  You store an SSH public key in SPIFFS on the ESP32 and no coded password is needed.  My wrapper around the example code from `libssh` is `ex_main`.  This just allows me to copy/paste the code as-is.  Is that enough information to get you going?

GeorgeFlorian, 2021-07-20:
> Yes, seems `exec` example is the closest to what you need to SSH to your server that connects to the ESP32 as AP. The `verify_knownhost` ensures you connect to only a server you already know. [...]

I will try it.
Thank you very much.

Isai-dc, 2022-10-09:
Thanks a lot for your response ewpa, I did what you said and it works. 

Now I am not that experienced in writing code so I am not really sure how to simplify your code. 

But I'm looking for a way to simplify your code to where you just have to put in the WiFi credentials, the server ip adres, the user, the password and the cmd code. 
Where then the serial output would just tell you if a connection has been made or not. 

In the end I want to make a physical enclosure for the esp32 with a couple buttons and LEDs to be able to turn my pc on / off.

ewpa, 2022-10-09:
Hi @Isai-dc, it may be easier to start with a simple WiFi example of your own and add in chunks of the example code gradually.