ewpa/LibSSH-ESP32 #13: Low download and high upload speed

VNovytskyi, 2021-11-05:
When I receive a file from the Ubuntu SCP server to ESP32 the size of receive package is too small (only 72 Bytes). When I transmit a file from ESP32 to the Ubuntu SCP server the size of the transmitting package is 4096 Byte (as a buffer which I set). In source files, I read that a function ssh_channel_read_timeout may return fewer bytes of data. How I can speed up the download?

ewpa, 2021-11-07:
As part of the porting process I reduce many windows and buffers.  The goal is to improve stability -- with stability being prioritised over performance.  Please could you enable verbose mode and review the output as hopefully this will provide further insight.  An example of how to achieve this is in example `libssh_scp.ino`, variable `verbosity` to at least 2:
```
static int verbosity = 2;
...
ssh_options_set(session, SSH_OPTIONS_LOG_VERBOSITY, &verbosity);
```
This should help find the limitation, e.g.
```
[1970/01/01 00:00:43.024618, 2] grow_window:  growing window (channel 43:0) to 72 bytes : not needed (72 bytes)
```
So it certainly seems to be attempting a window increase and being limited at 72 bytes.

ewpa, 2021-11-07:
On line 3737 of `src/channels.c` please change `WINDOWBASE` to `WINDOWBEGIN`.  Let me know if this fixes this for you and I will push the fix.

ewpa, 2021-11-08:
You found I gave the wrong line number.  Please change that line to the following.
```
uint32_t new_window = minimumsize > WINDOWBEGIN ? minimumsize : WINDOWBEGIN;
```
I have some regression tests to perform before I can release the code change.

ewpa, 2021-11-08:
Feel free to close the issue now.

closed 2021-11-08 by VNovytskyi