ewpa/LibSSH-ESP32 #14: System Event T not declared

bhboyle, 2021-12-17:
Hi there.

I was trying to compile the EXEC example this morning for the first time and came across the following error. 'system_event_t' has not been declared

I tried several different esp32 boards but they all had the same results. 

I am using Arduino 1.8.18 and 2.1.0 of the SSH library. 

Am I missing something?

Thanks for any help you may provide.
Cheers.

ewpa, 2021-12-17:
Please confirm the version of the arduino-esp32 in use in the Arduino boards manager.  This library and examples are tested against versions 1.0.6 and 2.0.0 only so far.

bhboyle, 2021-12-17:
I am running 2.0.0

Cheers.

bhboyle, 2021-12-17:
I just tried it on another machine and it worked so there is something wrong with my laptop. I will figure it out. 

Cheers.

bhboyle, 2021-12-17:
So if anyone else has this same issue here is what I had to do to fix it.

1. Uninstall Arduino
2. Delete Arduino folders
3. Clean out Librarys folder of ESP32 SSH
4. Clean out Hardware folder.
5. Reinstall Arduino
6. Add ESP32 back in boards manager
7. Add ESP32ssh back in Library manager
8. Test... it worked.

Cheers.

closed 2021-12-17 by ewpa