ewpa/LibSSH-ESP32 #15: Build error on ESP32Arduino 2.0.2

VNovytskyi, 2022-01-13:
Library LibSSH-ESP32 works successfully on ESP32Arduino 1.0.6. When I updated ESP32Arduino to the last version 2.0.2 occurs build error:
```
c:/users/vlad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Vlad\AppData\Local\Temp\arduino_build_490729\libraries\LibSSH-ESP32-2.1.0\LibSSH-ESP32-2.1.0.a(kex.c.o):(.literal.ssh_make_sessionid+0x28): undefined reference to `sha1'
c:/users/vlad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Vlad\AppData\Local\Temp\arduino_build_490729\libraries\LibSSH-ESP32-2.1.0\LibSSH-ESP32-2.1.0.a(kex.c.o): in function `ssh_make_sessionid':
C:\Users\Vlad\Documents\Arduino\libraries\LibSSH-ESP32-2.1.0\src/kex.c:1173: undefined reference to `sha1'
c:/users/vlad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Vlad\AppData\Local\Temp\arduino_build_490729\libraries\LibSSH-ESP32-2.1.0\LibSSH-ESP32-2.1.0.a(pki_mbedcrypto.c.o): in function `pki_sign_data':
C:\Users\Vlad\Documents\Arduino\libraries\LibSSH-ESP32-2.1.0\src/pki_mbedcrypto.c:1219: undefined reference to `sha1'
c:/users/vlad/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Vlad\AppData\Local\Temp\arduino_build_490729\libraries\LibSSH-ESP32-2.1.0\LibSSH-ESP32-2.1.0.a(pki_mbedcrypto.c.o): in function `pki_verify_data_signature':
C:\Users\Vlad\Documents\Arduino\libraries\LibSSH-ESP32-2.1.0\src/pki_mbedcrypto.c:1299: undefined reference to `sha1'
collect2.exe: error: ld returned 1 exit status
```
Then I tried to use ESP32Arduino 2.0.1 and have the same result. The last successful build is available on ESP32Arduino 2.0.0. I tried to find implementation for the ```sha1()``` function but failed.
I really want to use your beautiful library on the last ESP32Arduino 2.0.2, but I can`t find the source of the problem, I hope you help me!

ewpa, 2022-01-13:
Is your target the ESP32 or ESP32-C3?

VNovytskyi, 2022-01-14:
I switched to branch **github-issue-15** and successfully built the project on ESP32ArduinoCore **V2.0.2.** The library passed the download test in my application.
I tested several build use-cases:

| ESP32ArduinoCore  | LibSSH | Result |
| --------------------- | ------- | ------- |
|1.0.6 | 1.4.1|OK|
|1.0.6|2.1.0|OK|
|2.0.0|2.1.0|OK|
|2.0.2|2.1.0|Build error|
|2.0.2|issue-15|OK|

Thank you for fast replying to my problem! I hope you added this fix in the next release with backwards compatibility!

closed 2022-01-15 by ewpa