ewpa/LibSSH-ESP32 #17: Missing chacha20poly1305_cipher() ?

mnemo70, 2022-04-12:
Hello,
first of all thanks for your effort of bringing libssh to ESP32. I'm trying to integrate LibSSH-ESP32 into the [FujiNet project](https://github.com/FujiNetWIFI/fujinet-platformio), but I'm having a hard time. I can *almost* compile the project, but at the end the linker throws an error because ```ssh_get_chacha20poly1305_cipher``` is an undefined reference.

```c:/users/tm/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\fujinet-v1\libcca\libssh.a(libmbedcrypto.o):(.literal.ssh_crypto_init+0x24): undefined reference to `ssh_get_chacha20poly1305_cipher'
c:/users/tm/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\fujinet-v1\libcca\libssh.a(libmbedcrypto.o): in function `ssh_crypto_init':
D:\Dev\fujinet-platformio/lib/libssh/libmbedcrypto.c:1442: undefined reference to `ssh_get_chacha20poly1305_cipher'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\fujinet-v1\firmware.elf] Error 1```

There's the function definition in libssh/crypto.h:
```const struct ssh_cipher_struct *ssh_get_chacha20poly1305_cipher(void);```

But the package doesn't implement the function itself. The original libssh source implements it in [chachapoly.c](https://github.com/libssh/libssh-mirror/blob/master/src/chachapoly.c#L202), but that file is missing from the ESP32 port.

I have these libssh-related "build_flags" in platformio.ini, but it doesn't work.
```    -D ESP32=1
    -D LIBSSH_STATIC=1
    -D CONFIG_MBEDTLS_POLY1305_C=1
    -D CONFIG_MBEDTLS_CHACHA20_C=1
    -D CONFIG_MBEDTLS_CHACHAPOLY_C=1```

I'm just starting with Platform IO, so there might be much that I'm missing... Do you have any suggestions?

ewpa, 2022-04-13:
Those CONFIG_MBEDTLS settings are unset in the ESP32 Arduino SDK and hence would not be present when building any  project.  That is the reason that no port of those was attempted.  What was your goal with those?

mnemo70, 2022-04-13:
Thanks for getting back.

I added those settings in the hope that `ssh_get_chacha20poly1305_cipher()` would then be defined somewhere.

When I remove the MBEDTLS defines I get lots of errors like these:
```
c:/users/tm/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\fujinet-v1\libcca\libssh.a(libmbedcrypto.o):(.literal.chacha20_poly1305_cleanup+0x0): undefined reference to `mbedtls_chacha20_free'
c:/users/tm/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\fujinet-v1\libcca\libssh.a(libmbedcrypto.o):(.literal.chacha20_poly1305_cleanup+0x4): undefined reference to `mbedtls_poly1305_free'
c:/users/tm/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\fujinet-v1\libcca\libssh.a(libmbedcrypto.o):(.literal.chacha20_poly1305_set_iv+0xc): undefined reference to `mbedtls_chacha20_starts'
```
I forgot to mention that I also added the build_flag: `-D ESP32=1` because ESP32 is often referenced in the source, but was undefined.

When I remove the ESP32 define, `socketpair()` is not available:
```
lib/libssh/socket.c: In function 'ssh_socket_connect_proxycommand':
lib/libssh/socket.c:930:10: error: implicit declaration of function 'socketpair'; did you mean 'socket'? [-Werror=implicit-function-declaration]
    rc = socketpair(AF_UNIX, SOCK_STREAM, 0, pair);
```
Note: I am not including LibSSH-ESP32 as a PIO library, but added the directory to the main project in lib. When I tried it as a library, it resulted in #include chaos.

But where is `ssh_get_chacha20poly1305_cipher()`?

ewpa, 2022-04-14:
The above errors surrounding `chacha20_poly1305_set_iv` indicate that the `CONFIG_MBEDTLS` settings were defined, though not defined to `1`.  So I suspect there is something wrong/unusual with your PIO setup.  I do not have any experience with PIO.  I suggest you revisit adding as a library.  Perhaps the `#include` chaos was caused by the fact that libssh2 was already included in that project?

mnemo70, 2022-05-16:
Please close the issue, I don't think it's the right solution for esp-idf. Thank you!

closed 2022-05-18 by ewpa