ewpa/LibSSH-ESP32 #18: Reducing heap memory usage on the Embedded devices

VNovytskyi, 2022-04-14:
## Description
We are using the **SCP** feature in the project to download firmware from the remote server for the ESP32. Usually, our system has ~72 Kbytes free byte-access memory (total free memory ~108 KBytes). When we start to download the firmware file the free heap memory falls down to critical ~20KBytes and the system stops executing. It means that the SCP feature uses approximately 50 KBytes. 
For download files, via SCP we are using a buffer of 512 Bytes:
```c
#define SCP_CLIENT_TX_RX_BUFF 512

...

do {
  watchdog_reset_all();
  if (req_p->is_abort) {
    LOG_E(TAG, "Operation is aborted!\n");
    break;
  }
  
  int r = ssh_scp_read(src.scp, copy_buffer, SCP_CLIENT_TX_RX_BUFF);
  if (r == SSH_ERROR) {
    LOG_E(TAG, "Error reading scp: %s\n", ssh_get_error(src.session));
    break;
  } else if (r == 0) {
    LOG_E(TAG, "Read 0 bytes.\n");
    break;
}

  if (dest.file.write(copy_buffer, r) <= 0) {
    LOG_E(TAG, "Error writing in local file: %s\n", strerror(errno));
    break;
   }

  transferred_bytes_count += r;
  req_p->progress = transferred_bytes_count * 100 / file_size;

} while (transferred_bytes_count < file_size);
```

Also, I am testing with the WINDOWBEGIN macro (in _channels.c_):
```c
//#define WINDOWBEGIN 8000
#define WINDOWBEGIN 1000
```

If using `WINDOWBEGIN 1000` then the low value of the free heap is ~45 KBytes, that's better. Are there some buffers and settings that we can change to reduce heap memory usage? It will be beneficial to continue using your library for current and future projects.

## Environment
_Device:_ ESP32
_Arduino core:_ 2.0.2
_LibSSH:_ 2.2.0

ewpa, 2022-04-14:
Review the version of esp32-arduino in use.  Early 2.0+ versions used much more RAM than 1.0.6 (say).  You may also wish to reduce these settings further than I did:

- `BUFFER_SIZE_MAX` reduced from 0x10000000 to 16384 in `buffer.c`
- `WINDOWBASE` and `WINDOWBEGIN` reduced from 1280000 to 8000 in `channels.c`
- `CHANNEL_INITIAL_WINDOW `reduced from 64000 to 250 in `channels.c`
- `local_maxpacket` reduced from 35000 to 32768 in `messages.c`
- `local_window` reduced from 32000 to 8000 in `messages.c`

VNovytskyi, 2022-04-15:
Ok, I will try it, thank you for your answer!

closed 2022-04-15 by VNovytskyi