ewpa/LibSSH-ESP32 #19: Which comands i should use?

AposazhennikovDS, 2022-06-08:
Greetengs sir! Thank you for your work! Im trieng to make SSH client on ESP32, my point is to connect my esp to server on linux and exec some comands on linux server. But then im trieng to use some default commands i get some errors, please could you give me link to actual `documentation?``
[image]
`

ewpa, 2022-06-08:
Hello, I'm missing the first error in your compilation which may hint to the issue.  Have you compiled and run the examples successfully?

AposazhennikovDS, 2022-06-09:
Thank you for your quick response, im just trying to make simply clien, and the examples are too complicated for me to understand, so, when i simply replaced "Your ssid"/ "localhost" and other vars to my vars it doesnt compilated because of error. I tried to use examples which named "exec" and OTAclientSCP(exec doesnt work; Otaclient is working).  it would be great if i can wrote code by myself. So maybe you will help me to understand why in my code i gettings error like that ": error: 'SSH_OPTIONS_HOST' was not declared in this scope
   ssh_options_set(my_ssh_session, SSH_OPTIONS_HOST, ****)"

ewpa, 2022-06-09:
Yes, the examples were ported mainly from upstream where they were written for Unix/Linux type systems.  Hence for SSH_OPTIONS_HOST you may need to include additional headers, for example in `exec.ino`:
```
#include "libssh/priv.h"
#include <libssh/libssh.h>
#include "examples_common.h"
```

AposazhennikovDS, 2022-06-09:
Thank you sir! I also have a question about the algorithm of executing  commands at the server. Am I to understand that firstly I must  create two necessary objects "channel" and "session" if i want to send the commands? Also when i oppened channel can i send 2 or more(unlimited) commands to server? Or we have limit?

closed 2025-04-06 by ewpa