ewpa/LibSSH-ESP32 #20: Authorization over id_rsa key --> File not Found at littleFS

trebur, 2022-08-03:
I'm try you remote exec sketch but i get always file not found
here a directory listing
Directory-Listing: /spiffs/.ssh
  FILE: /spiffs/.ssh/id_rsa.pub
  FILE: /spiffs/.ssh/id_rsa 
  FILE: /spiffs/.ssh/known_hosts  

but ....
1970/01/01 00:00:04.943585, 1] ssh_known_hosts_read_entries:  Failed to open the known_hosts file '/spiffs/.ssh/known_hosts': No such file or directory
[1970/01/01 00:00:04.956632, 1] ssh_known_hosts_read_entries:  Failed to open the known_hosts file '/etc/ssh/ssh_known_hosts': No such file or directory
.....
[1970/01/01 00:00:05.798117, 3] ssh_userauth_publickey_auto:  Trying to authenticate with /spiffs/.ssh/id_rsa
[1970/01/01 00:00:05.807849, 1] ssh_pki_import_pubkey_file:  Error opening /spiffs/.ssh/id_rsa.pub: No such file or directory
[1970/01/01 00:00:05.818864, 1] ssh_pki_import_privkey_file:  Error opening /spiffs/.ssh/id_rsa: No such file or directory
[1970/01/01 00:00:05.829540, 3] ssh_userauth_publickey_auto:  Private key /spiffs/.ssh/id_rsa doesn't exist.
[1970/01/01 00:00:05.839095, 2] ssh_userauth_publickey_auto:  Tried every public key, none matched

the only difference are  use littleFS is that a Problem ?
I get the same result if i use SSH_OPTIONS_SSH_DIR option (also with ~) .
Any idea ?

ewpa, 2022-08-04:
Hi, please could you share a fork which includes the changes you made to use LittleFS versus SPIFFS.

playmiel, 2022-11-24:
with some research i think the problem is libssh related or a conflict with spiffs if you want to fix the problem remove verfify_knownhost