ewpa/LibSSH-ESP32 #27: Reverse SHH tunnel [ QUESTION ]

zekageri, 2023-06-01:
Hi! 

My goal is to reach my esp32 from outside of my network without port forwarding.
Right now my solution is the following:

ESP32 inside local network
Node.js server is outside on the internet.

ESP32 connects to a node.js server via raw websockets.
Client loads an html page from the node.js server
ESP32 forwards every message to the node js server
Node js server forwards every message from it's clients to the esp32 and reverse.

This requires duplicate codes because the server has to contain the same HTML page as the esp32 and every front end script should exists twice.

I wonder if it is possible to forward every HTTP and websocket request from the server to my esp32 which is inside a local network.
I want to be able to reach my esp32 from outside of the network without port forwarding just like i would reach it from within the network. The esp32 hosts a webserver and serves html pages.

Thank you for your response.

ewpa, 2023-06-01:
Hello!  Your main constraint is not wanting to use port forwarding.  To completely avoid port forwarding you would need a publicly routable IP address on your ESP32.  The method to achieve this is to assign a public IPv6 address to the ESP32.  Another option would be to create a VPN between the external and internal server networks and run a web proxy on the node.js server to forward/reverse all HTTP(S) traffic.

zekageri, 2023-06-01:
Oh I see. Thank you for your help. I appreciate.

closed 2023-06-01 by zekageri