ewpa/LibSSH-ESP32 #32: How to create SSH server using LibSSH-ESP32?

Akshat-Statcon, 2024-02-08:
I want to create ssh server on ESP32 and want to connect other devices through it and communicate using the server.  
I am trying to create a server using libssh library but can't set private keys in the bind options.

I'm using the following line to set the private key and private_key contains the path to the ECDSA key.
ssh_bind_options_set(pSshBind, SSH_BIND_OPTIONS_ECDSAKEY, private_key);

But when I run the following lines of code 

if (ssh_bind_listen(pSshBind) < 0) {
        fprintf(stderr, "Error accepting a connection : %s\n", ssh_get_error(pSshBind));
        ssh_bind_free(pSshBind);
        return;
    }

I get this error -  Error accepting a connection : Failed to import private ECDSA host key.

Could someone please resolve this error or instruct on how to create a server from scratch?

ewpa, 2024-02-08:
You will need to set the host key, e.g.
```
#ifndef SSHD_KEYDIR
#define SSHD_KEYDIR "/spiffs"
#endif
#ifndef SSHD_KEYFILE
#define SSHD_KEYFILE "/ided25519"
#endif
[...]
ssh_bind_options_set(sshbind, SSH_BIND_OPTIONS_HOSTKEY, SSHD_KEYDIR SSHD_KEYFILE);
```

Akshat-Statcon, 2024-02-14:
Thank you for your response! Unfortunately, the solution provided earlier did not resolve the issue for me. I also attempted to generate new keys using the code provided in the example you shared from GitHub (https://github.com/ewpa/LibSSH-ESP32/blob/master/examples/samplesshd-kbdint/samplesshd-kbdint.ino), but I am still encountering the same problem.
[image]

ewpa, 2024-02-14:
Please do not convert text into a picture since it makes reading and searching very difficult.  Which example did you use to create the host key and what was the output?

Akshat-Statcon, 2024-02-15:
Sorry for the confusion, 
1. I used the [samplesshd-kbdint.ino](https://github.com/ewpa/LibSSH-ESP32/blob/master/examples/samplesshd-kbdint/samplesshd-kbdint.ino) to generate the host keys
2. The following t=is the output for that code
`09:31:40.504 -> ets Jul 29 2019 12:21:46
09:31:40.504 -> 
09:31:40.504 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
09:31:40.541 -> configsip: 0, SPIWP:0xee
09:31:40.541 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
09:31:40.541 -> mode:DIO, clock div:1
09:31:40.541 -> load:0x3fff0030,len:1184
09:31:40.541 -> load:0x40078000,len:13260
09:31:40.541 -> load:0x40080400,len:3028
09:31:40.541 -> entry 0x400805e4
09:31:40.944 -> E (11) SPIFFS: mount failed, -10025
09:31:40.944 -> % No formatted SPIFFS filesystem found to mount.
09:31:40.944 -> % Format SPIFFS and mount now (NB. may cause data loss) [y/n]?
09:36:44.965 -> % Formatting...
09:36:49.527 -> % Mounted SPIFFS used=0 total=1318001

09:36:49.673 -> % WiFi enabled with SSID=S-INDIA
09:36:55.354 -> % WiFi connected
09:36:59.430 -> % IPv4 Address: 192.168.20.172
09:37:00.899 -> % IPv6 Address: fe80:0000:0000:0000:26dc:c3ff:fed1:49e0
09:37:05.402 -> % Timeout waiting for all IP addresses
09:37:05.402 -> % Execution in progress: samplesshd-kbdint --hostkey /spiffs/.ssh/id_ed25519 ::
09:37:05.402 -> [SNIP STDOUT START]
09:37:05.481 -> Error listening to socket: Failed to import private RSA host key
09:37:05.526 -> [SNIP STDOUT FINISH]
09:37:05.526 -> % Execution completed: rc=1, elapsed=95ms`

ewpa, 2024-02-15:
Please use the ```keygen2``` example to create the host key.

closed 2025-04-06 by ewpa