ewpa/LibSSH-ESP32 #36: Waiting for an incoming connection in non-blocking mode

VNovytskyi, 2025-02-03:
**Is there a way to wait for an incoming connection in non-blocking mode?** I tried to implement an alternative method like _**ssh_bind_accept_alt**_, but it is not working. Maybe I missed some information in your code or docs? The main goal of this feature is to have the ability to destroy the SSH server and release occupied memory in each moment of time (depending on the system configuration).
```c
static int ssh_bind_accept_alt(ssh_bind ssh_bind, ssh_session ssh_session) {
  // Params check
  if ((ssh_bind == NULL) || (ssh_session == NULL)) {
    LOG_E(TAG, "Wrong params! (%p, %p)\n", ssh_bind, ssh_session);
    return SSH_ERROR;
  }

  // Get file descriptor via LibSSH public API
  socket_t bind_fd = ssh_bind_get_fd(ssh_bind);
  if (bind_fd < 0) {
    LOG_E(TAG, "Can't accept new clients on a not bound socket: %d!\n", bind_fd);
    return SSH_ERROR;
  }

#if 0
  // Set non-blocking mode
  int fd_flags = fcntl(bind_fd, F_GETFL, 0);
  if (fd_flags < 0) {
    LOG_E(TAG, "Fail to get flags of file descriptor: %d!\n", fd_flags);
    return SSH_ERROR;
  }
  int fd_res = fcntl(bind_fd, F_SETFL, fd_flags | O_NONBLOCK);
  if (fd_res < 0) {
    LOG_E(TAG, "Fail to set flags of file descriptor: %d!\n", fd_res);
    return SSH_ERROR;
  }
#endif

#if 1
  // Prepare timeout structure
  struct timeval timeout = {1, 0};
  fd_set fds;
  FD_ZERO(&fds);
  FD_SET(bind_fd, &fds);

  // Waiting for incoming connection
  LOG_I(TAG, "Waiting for incoming connection...\n");
  int ready = 0;
  while (true) {
    if (ssh_serv.mute) {
      // Received external signal to break operation
      LOG_I(TAG, "Detected mute state. Break accept.\n");
      close(bind_fd);
      return SSH_ERROR;
    }

    // Execute socket select
    ready = select(bind_fd + 1, &fds, NULL, NULL, &timeout);

    if (ready > 0) {
      // Accepted
      LOG_I(TAG, "Accepted.\n");
      break;
    } else if (ready < 0) {
      // Socket select error
      LOG_E(TAG, "Fail to select socket: %d!\n", ready);
      return SSH_ERROR;
    } else if (ready == 0) {
      LOG_I(TAG, "Timeout: No connection received\n");
    }

    utils_delay(1000);
  }

#endif

  // Accept connection
  int client_fd = accept(bind_fd, NULL, NULL);
  if (client_fd == SSH_INVALID_SOCKET) {
    if (errno == EINTR) {
      LOG_E(TAG, "Accepting a new connection (child signal error): %d\n", client_fd);
    } else {
      LOG_E(TAG, "Accepting a new connection: %d\n", client_fd);
    }
    return SSH_ERROR;
  }

  // Call internal API to accept SSH connection
  int rc = ssh_bind_accept_fd(ssh_bind, ssh_session, client_fd);
  if (rc == SSH_ERROR) {
    close(client_fd);
  }

  // Return result of the internal operation
  return rc;
}
```
I am waiting for a SSH client by:
```c
r = ssh_bind_accept(ssh_serv.sshbind, ssh_serv.session);
```
But it is a blocking method and the user needs to use the separated thread.
I used the MobaXterm and PuTTy terminals and they work very well with your library in blocking mode.

ewpa, 2025-02-03:
You limitation appears to be the requirement to use `ssh_bind` in all cases, setting options before listening, e.g.
  1. `ssh_bind_new`
  2. `ssh_bind_options_set` ...
  3. `ssh_bind_listen`
  4. `accept(ssh_bind_get_fd(sshbind), ...`
  5. `ssh_bind_accept_fd`
  6. Send the `fd` to a thread and resume that thread.
  7. Back to (4).

But do you want to call `ssh_bind_free`?  You would writing an `inetd` and spinning up a new bind per incoming connection in that case.

VNovytskyi, 2025-02-03:
First of all, I am looking for the way to call _**ssh_bind_accept**_ method in non-blocking mode. Because this method is blocked thread and wait forever for client. Something like that:
```c
while(true) {
    if (!ssh_bind_accept(...)) {
        // No client
        // Some code here
        vTaskDelay(...);
        continue;
    }
    
    // Handle SSH client
}
```

ewpa, 2025-02-03:
Your approach looks textbook according to *Stevens, 1990*.  Hence on the face of it I cannot see the error.  What is the behavior of the code not working?  Do you have the code on Linux where it can be more easily debugged?

VNovytskyi, 2025-02-04:
So, I have found the solution to my issue. 😎

<h1>Issue explanation</h1>

First, I'd like to explain my issue again: my goal was to call `ssh_bind_accept` in non-blocking mode. 
But this method is using the lwIP `accept(...)` call which is blocking by default. 
I want to use `ssh_bind_accept` in the next way:
```c
  // accept incomming connections
  while (true) {
    // Delay to reduce load on CPU and reset system watchdog
    utils_delay(1000);

    if (mute) {
      LOG_I(TAG, "Detected mute state.\n");
      goto END;
    }

    // Non-blocking method
    r = ssh_bind_accept(sshbind, session);

    // Handle result
    if (r == SSH_OK) {
      // Connection accepted
      break;
    } else if (r == EAGAIN) {
      // No connection
      continue;
    } else {
      // Failed
      LOG_E(TAG, "Error accepting a connection: %d, %s\n", r, ssh_get_error(ssh_serv.sshbind));
      goto END;
    }
  }
```
instead of:
```c
  LOG_I(TAG, "Started ssh server on port %d\n", port);

  // accept incomming connections (blocking function!)
  r = ssh_bind_accept(sshbind, session);
  if (r == SSH_ERROR) {
    LOG_E(TAG, "Error accepting a connection: %s\n", ssh_get_error(ssh_serv.sshbind));
    goto END;
  }
  is_connected = 1;
```

<h1>Solutions</h1>

There are several possible solutions to solve this issue:
1. Create a lwIP socket manually using the O_NONBLOCK option, wait for connection and then call `ssh_bind_accept_fd`
```c
sock_local = socket(AF_INET, SOCK_STREAM, 0)
fcntl(sock_local, F_SETFL, O_NONBLOCK)
setsockopt(sock_local, SOL_SOCKET, SO_REUSEADDR, ...)
bind(...)
listen(sock_local, 1)
 while (true) {
    if (mute) {
      LOG_I(TAG, "Detected mute state.\n");
      goto END;
    }

    client_socket = accept(sock_local, NULL, NULL);
    if (client_socket > 0) {
      LOG_I(TAG, "Accepted!!! (%d, %d)\n", sock_local, client_socket);
      break;
    }

    if ((errno != EAGAIN) && (errno != EWOULDBLOCK)) {
      LOG_E(TAG, "Accepting a new connection: %d, %d\n", sock_local, errno);
    }

    utils_delay(1000);
  }

ssh_bind_options_set(sshbind, SSH_BIND_OPTIONS_IMPORT_KEY, key);

ssh_bind_accept_fd(sshbind, session, client_socket);
```
2. Make minor modifications to the LibSSH sources:
Modify src/bind.c::bind_socket (add O_NONBLOCK option):
```c
static socket_t bind_socket(ssh_bind sshbind, const char *hostname,
    int port) {
    char port_c[6];
    struct addrinfo *ai;
    struct addrinfo hints;
    int opt = 1;
    socket_t s;
    int rc;
    char err_msg[SSH_ERRNO_MSG_MAX] = {0};

    ZERO_STRUCT(hints);

    hints.ai_flags = AI_PASSIVE;
    hints.ai_socktype = SOCK_STREAM;

    snprintf(port_c, 6, "%d", port);
    rc = getaddrinfo(hostname, port_c, &hints, &ai);
    if (rc != 0) {
        ssh_set_error(sshbind,
                      SSH_FATAL,
                      "Resolving %s: %d", hostname, rc);
        return -1;
    }

    s = socket (ai->ai_family,
                           ai->ai_socktype,
                           ai->ai_protocol);
    if (s == SSH_INVALID_SOCKET) {
        ssh_set_error(sshbind, SSH_FATAL, "%s",
                      ssh_strerror(errno, err_msg, SSH_ERRNO_MSG_MAX));
        freeaddrinfo (ai);
        return -1;
    }

    if ((rc = fcntl(s, F_SETFL, O_NONBLOCK)) < 0) {
        ssh_set_error(sshbind, SSH_FATAL, "Can't set NONBLOCK mode for server socket.");
        return -1;
    }

    if (setsockopt(s, SOL_SOCKET, SO_REUSEADDR,
                   (char *)&opt, sizeof(opt)) < 0) {
        ssh_set_error(sshbind,
                      SSH_FATAL,
                      "Setting socket options failed: %s",
                      ssh_strerror(errno, err_msg, SSH_ERRNO_MSG_MAX));
        freeaddrinfo (ai);
        CLOSE_SOCKET(s);
        return -1;
    }

    if (bind(s, ai->ai_addr, ai->ai_addrlen) != 0) {
        ssh_set_error(sshbind,
                      SSH_FATAL,
                      "Binding to %s:%d: %s",
                      hostname,
                      port,
                      ssh_strerror(errno, err_msg, SSH_ERRNO_MSG_MAX));
        freeaddrinfo (ai);
        CLOSE_SOCKET(s);
        return -1;
    }

    freeaddrinfo (ai);
    return s;
}
```
Modify src/bind.c::ssh_bind_accept (add while loop for lwIP::accept):
```c
int ssh_bind_accept(ssh_bind sshbind, ssh_session session)
{
    socket_t fd = SSH_INVALID_SOCKET;
    int rc;

    if (sshbind->bindfd == SSH_INVALID_SOCKET) {
        ssh_set_error(sshbind, SSH_FATAL,
                      "Can't accept new clients on a not bound socket.");
        return SSH_ERROR;
    }

    if (session == NULL) {
        ssh_set_error(sshbind, SSH_FATAL, "session is null");
        return SSH_ERROR;
    }

    while (true) {
      fd = accept(sshbind->bindfd, NULL, NULL);

      if (fd > 0) {
          break;
      }

      if ((errno == EWOULDBLOCK) || (errno == EAGAIN)) {
        return EAGAIN;
      }

      if (fd == SSH_INVALID_SOCKET) {
          char err_msg[SSH_ERRNO_MSG_MAX] = {0};
          if (errno == EINTR) {
              ssh_set_error(sshbind, SSH_EINTR,
                            "Accepting a new connection (child signal error): %s",
                            ssh_strerror(errno, err_msg, SSH_ERRNO_MSG_MAX));
          } else {
              ssh_set_error(sshbind, SSH_FATAL,
                            "Accepting a new connection: %s",
                            ssh_strerror(errno, err_msg, SSH_ERRNO_MSG_MAX));
          }
          return SSH_ERROR;
      }
    }

    rc = ssh_bind_accept_fd(sshbind, session, fd);

    if (rc == SSH_ERROR) {
        CLOSE_SOCKET(fd);
        ssh_socket_free(session->socket);
    }

    return rc;
}
```

<h1>Conclusion</h1>

Using the second solution (modify LibSSH) I can use the code mentioned in the "Issue explanation" chapter. This allows me to break the execution of my SSH server and release all dynamic memory in ESP32.

It may be a useful feature in your library.

ewpa, 2025-02-04:
I saw your disabled code which reinforced your requirement.  I am pleased you found a solution.  I prefer #1 :-)  since I hesitate to add features not in upstream libbsh.

closed 2025-02-04 by ewpa