ewpa/LibSSH-ESP32 #37: ArduinoIDE can't compile any example.

Jullyfish, 2025-04-01:
Any example gives an error:
```
/home/user/Arduino/libraries/LibSSH-ESP32/src/misc.c:1988:8: error: assignment to 'int' from 'char *' makes integer from pointer without a cast [-Wint-conversion]
 1988 |     rv = strerror_r(err_num, buf, buflen);
```

The simplest sketch:
```CPP
#include <libssh_esp32.h>

void setup()
{
	libssh_begin();
}

void loop()
{

}
```

Jullyfish, 2025-04-01:
Okay, I just add casting, now it compiles:
```CPP
rv = (int)strerror_r(err_num, buf, buflen);
```

ewpa, 2025-04-01:
You should not have to do that.  What versions of IDE and ESP32-Arduino boards are you using?

Jullyfish, 2025-04-01:
> What versions of IDE
```
Version: 2.3.4
Date: 2024-12-03T10:51:12.539Z
CLI Version: 1.1.1

Copyright © 2025 Arduino SA
```

> ESP32-Arduino boards

esp-wroom-32 38 pins.

ewpa, 2025-04-01:
Thanks, also what is the version of the ESP32 Arduino core software?

Jullyfish, 2025-04-01:
I'm not really sure how to check the version. :^)
```CPP
Serial.println(ESP.getCoreVersion());
=>
3.2.0
```

ewpa, 2025-04-01:
3.0.x is the latest version I have built and tested against.  Confirming FTBFS with core 3.2.0.

Jullyfish, 2025-04-05:
Fixed in [c2ef948](https://github.com/ewpa/LibSSH-ESP32/commit/c2ef948b0f1712ba1e4e6724c1062b1e4a99e370).

closed 2025-04-05 by Jullyfish