ewpa/LibSSH-ESP32 #38: OTA scp example.

Jullyfish, 2025-04-02:
scp cannot copy file from esp32 approximately more than 13000B, gives error:
```
ota% ssh_scp_read error: Socket error: disconnected
```

I  searched in LisSSH-ESP32 what place returns error code. It's a function:
```C
/* poll.c */
int ssh_poll_ctx_dopoll(ssh_poll_ctx ctx, int timeout) 
    /* ... */
            p->lock_cnt++;                                                      
            if (p->cb && (ret = p->cb(p, fd, revents, p->cb_data)) < 0) {       
                 if (ret == -2) {                                                                                                               
                     // Here!
                     return -1;                                            
                 }   
    /* ... */
```

ArduinoIDE:
```
Version: 2.3.4
Date: 2024-12-03T10:51:12.539Z
CLI Version: 1.1.1

Copyright © 2025 Arduino SA
```
board: esp-wroom-32 38 pins.
ESP Arduino core: 3.2.0

Jullyfish, 2025-04-02:
`libssh_scp` example has the same problem.

Jullyfish, 2025-04-03:
Okay, I played with it and found error and can OTA update after all. The error is:
```
[1970/01/01 00:01:11.336700, 3] channel_open:  Creating a channel 43 with 2097152 window and 32768 max packet
```

I don't know from where it takes 32768 size. I hardcoded it in channel_open() to buffer size and it worked.

ewpa, 2025-04-03:
I had not recreated your issue yet.  The 2MB window size looked large for an ESP32.  Which version of the library are you using and do you apply any patches?

Jullyfish, 2025-04-03:
> Which version of the library are you using

The last one: 5.4.0

> do you apply any patches?

Some debug print and [this](https://github.com/ewpa/LibSSH-ESP32/issues/37).

> I don't know from where it takes 32768 size.

> The 2MB window size looked large for an ESP32. 

First, I found a macro `CHANNEL_MAX_PACKET` in `channel.c`. It define 32768 (not found first time, got a typo with `grep`). And after this macro the next one is:
```
#define WINDOW_DEFAULT (64*CHANNEL_MAX_PACKET)
```
That's why window is big. (And what is the "window" by the way?)

ewpa, 2025-04-03:
Thank you.  I believe the `WINDOW` constant names in `channels.c` changed in libssh 0.11 commit 60db50805 and this was not properly reflected in library version 5.x.x.

> channel: use a larger window size to increase receive throughput.
> This changes the default window from 1.28MB to 2MiB.

For reference the window is a variable size buffer and used to increase communication efficiency automatically as connection speeds change up or down.  For the ESP32 port I reduce this since the ESP32 has very small RAM.

ewpa, 2025-04-05:
@Jullyfish please test branch github-issue-38, commit c2ef948 (potential 5.5.0 release) and let me know if this resolves these issues.

Jullyfish, 2025-04-05:
Yeah, it works. Thank you very much!

closed 2025-04-05 by Jullyfish